Pitmairen/bryton-gps-linux #25: OSX support

wiedo, 2015-05-30:
Not an issue but just a note, this script also works (partly) on OSX. Only the GPS correction isn't working but the strava uploading also works on my Bryton Rider 21, great! The bryton app is terrible!

This is the terminal command i run:
python brytongps.py --device /Volumes/BRYTON/ --fspath /Volumes/BRYTON/ -S ~/Desktop/bryton/ --fake-garmin -T 0 --tcx

For strava support you have to run:
sudo easy_install mechanize

Pitmairen, 2015-05-30:
Hi, that's cool. Most of the code should work on OS X and also even windows, at least for newer devices with a normal file system. For older devices like the rider 40 code for sending the SCSI commands will be needed before it will work. 

So it actually works and you can upload the trips to strava?
What do you mean by gps correction?

A tip to simplify the command you can create a bash alias or a bash script so that you don't need to type all the paths every time.

wiedo, 2015-05-30:
I meant the elevation correction DB isn't working. The rest is working
perfectly fine. The bash thing is the first thing I'm going to do. Maybe
even automatically run it when I connect my Bryton. Love it, thanks for
this 👍!

Pitmairen, 2015-05-30:
Ok, I see. If the GDAL library (https://pypi.python.org/pypi/GDAL/) works on osx it should probably work, but i haven't tested the elevation db option in quite a while so it may not even work on linux. But I will check to see if it needs to be updated.

Pitmairen, 2015-06-01:
The elevation db code still works on linux. But the gdal library may be a little fiddly to install. But there are some packages for OSX available here: http://www.kyngchaos.com/software:frameworks (I haven't tested these, so i don't know if they work.)

There is also some northern parts of the globe that is not available in the db. Here is a map that shows the available locations http://srtm.csi.cgiar.org/SELECTION/inputCoord.asp