Pitmairen/bryton-gps-linux #26: Bryton Rider 310

vagrale13, 2016-05-25:
Is it possible to support **Bryton Rider 310** ?
The device works fine with nautilus.
All routes in device are in _.fit_ type files.
```
:~$ lsusb
0483:5720 SGS Thomson Microelectronics
```
Trying with the command with all old devices if that helps
`sudo python ./brytongps.py --storage`

and the outputs

_Rider40_ 
```
Traceback (most recent call last):
  File "brytongps.py", line 480, in <module>
    sys.exit(main())
  File "brytongps.py", line 349, in main
    module, device = get_device(dev_access)
TypeError: 'NoneType' object is not iterable
```

_Rider20+ (works with rider21)_
```
Traceback (most recent call last):
  File "./brytongps.py", line 482, in <module>
    sys.exit(main())
  File "./brytongps.py", line 375, in main
    module, device = get_device(dev_access)
TypeError: 'NoneType' object is not iterable
```

_Rider20 (works with cardio35)_
`Error:  Unknown Device`

_Rider35_
```
Traceback (most recent call last):
  File "./brytongps.py", line 454, in <module>
    sys.exit(main())
  File "./brytongps.py", line 348, in main
    module, device = get_device(dev_access)
TypeError: 'NoneType' object is not iterable
```

_Rider50_
```
Traceback (most recent call last):
  File "./brytongps.py", line 483, in <module>
    sys.exit(main())
  File "./brytongps.py", line 376, in main
    module, device = get_device(dev_access)
TypeError: 'NoneType' object is not iterable
```

Pitmairen, 2016-05-25:
The older devices, which this software supports, use a different file format that is not readable by other applications and can not be directly uploaded to strava and similar sites. The point of this software is to extract the data from these device and convert it to standard formats so it can be used by other applications. 

Newer devices Rider 310 seems to have switched to use the .fit format which should be usable by most other application. So then this software is not really that useful for these newer devices. 

But still, since they now are using the .fit format, it should not be that hard to add support for this device. If we can just parse the fit files, then it should be easy to support this device. I don't have time to do it right now, but maybe in a couple of weeks. 

If you can upload a zip with all the content that is shown in nautilus when you connect the device, then I have some data to work with.

vagrale13, 2016-05-25:
Ok i understand. Here is the BRYTON folder
[BRYTON.zip](https://github.com/Pitmairen/bryton-gps-linux/files/283028/BRYTON.zip

Pitmairen, 2016-06-04:
I have created a branch for Rider 310 now.
It is at least working with the data that was in you zip file. I'm not sure if the time stamps are correct and there may be issues with different units (m/s, kph, etc.). I haven't tested it that much.

It can't currently auto detect the device so you have to specify the following options to make it work:
```
brytongps.py -D something -FS /path/to/BRYTON/

vagrale13, 2016-06-04:
Here is the ouput
```
:~$ sudo python ./brytongps.py -D --storage -FS /media/BRYTON/
Traceback (most recent call last):
  File "./brytongps.py", line 33, in <module>
    import rider310
  File "/bryton-gps-linux-rider310/code/rider310.py", line 27, in <module>
    import fitparse
  File "/bryton-gps-linux-rider310/code/fitparse/__init__.py", line 1, in <module>
    from fitparse.base import FitFile, FitParseError
  File "/bryton-gps-linux-rider310/code/fitparse/base.py", line 5, in <module>
    from fitparse.profile import FIELD_TYPE_TIMESTAMP, MESSAGE_TYPES
  File "/bryton-gps-linux-rider310/code/fitparse/profile.py", line 7, in <module>
    from fitparse.records import (
  File "/bryton-gps-linux-rider310/code/fitparse/records.py", line 3, in <module>
    from six.moves import zip_longest
ImportError: cannot import name zip_longest
```

Pitmairen, 2016-06-04:
Are you using python2 or python3?

I have only tested this with python2, i'm not sure it works with python3.

Pitmairen, 2016-06-04:
Ok, so I don't think the problem is the python version, but you will have to use python2. I think the problem is that your python install for some reason loads an old version of the "six" module. 

If you start python in the terminal and type this:
```
import six
print six.__version__
```
what is the output?

vagrale13, 2016-06-04:
I'm using Python 2.7.3.
and the output
```
>>> import six
>>> print six.__version__
1.1.0
```

Pitmairen, 2016-06-04:
Ok, so that is a very old version, around 5 years old.

You should install the latest version 1.10.0. Either using you distro's package manager or using pip:
```
pip install six

vagrale13, 2016-06-04:
Ok, i upgrade 'six', and now run without these errors.
But no results yet
```
:~$ sudo python ./brytongps.py -D --storage -FS /media/BRYTON/
usage: brytongps.py [-h] [--device DEVICE] [--fspath FSPATH] [--list-history]
                    [--tracks TRACKS [TRACKS ...]] [--summary] [--gpx]
                    [--gpxx] [--tcx] [--json] [--save-to SAVE_TO]
                    [--out-name OUT_NAME] [--no-whitespace] [--strava]
                    [--strava-email [STRAVA_EMAIL]]
                    [--strava-password [STRAVA_PASSWORD]] [--fake-garmin]
                    [--fix-elevation [N]] [--strip-elevation]
                    [--use-elevation-db] [--storage] [--no-laps]
brytongps.py: error: argument --device/-D: expected one argument
```

Pitmairen, 2016-06-05:
You are missing the argument to the -D option. Try:
```
python ./brytongps.py -D abc -FS /media/BRYTON/ -L
```
The D option is not used for your device so it doesn't matter what the value is, I just used abc.

Also I don't think you need sudo for your device. It is needed for some other devices, but it shouldn't be needed for your device.

vagrale13, 2016-06-05:
You have right. Now works fine!
Thanks for help and for the 'bryton-gps-linux'

closed 2016-06-05 by vagrale13