Pitmairen/bryton-gps-linux #27: error Permission denied

mrredpin, 2017-07-07:
Error:  Failed to open device "/dev/disk/by-id/usb-BRYTON_MASS_STORAGE_000000000001-0:0" (Permission denied).
got this error 
is it right that that there is only SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5720", GROUP="users" in the udev rule 
and i had to creat the file?
im using a rider 20

Pitmairen, 2017-07-08:
Try to run the following command in the terminal, that should list all the groups you are a member of:
```
groups
```
And then replace the GROUP="users" with a group with one returned by the above command. 

Alternatively you can add OWNER="myusername" to the rule and set it to your own username. 

Hopefully this should work. :)

mrredpin, 2017-07-14:
thx that worked fine :)