Pitmairen/bryton-gps-linux #28: Rider35 GPS sat pos updates

rhy-ama, 2021-02-06:
Hello! Is it possible to upload GPS updates to these legacy devices ?

Pitmairen, 2021-02-06:
What do you mean by GPS updates?

Anyway this project has no support for writing. It can only read data from the devices.

rhy-ama, 2021-02-06:
There are satellite position updates published in the industry standard format xml (epehemeris data to speed up satellite fix). I thought maybe you figured out the way to mount write access for updates. Thank you for replying.

Pitmairen, 2021-02-06:
I still have my device, but I'm not using it that much anymore. So I doubt this project will ever get write support, for these kind of updates or anything else, unless someone else feels like figuring out how to write to the devices without breaking them.